TLeconte/acarsdec #8: Decoding from RTL-SDR breaks after ~10 messages

jskrien, 2017-01-27:
While decoding from the RTL-SDR, the first 10 messages are read reliably. After this point, I see more and more "too many parity errors" until eventually all messages are parity errors. Restarting acarsdec fixes the issue for another ~10 messages. This issue does not seem to exist in the 3.2 version. I have not been able to find the issue in the source.

ckuethe, 2017-02-13:
Have you tried a git bisection to see which commit introduced this behavior? https://git-scm.com/docs/git-bisect

ckuethe, 2017-02-22:
I can't reproduce this: camping outside my local airport I don't see any parity errors.

Additionally, after replaying captured traffic while adding increasing amounts of noise I can trigger parity errors but the decoder continues to work until the signal is completely buried.

DickvanNoort, 2017-04-08:
I am running the latest version (34) without any problems 24/7 from a RPi 2B.
You might want to make sure that the dongle is getting enough power.
I have seen similar problems when the dongle is connected to a power supply which was not able to supply enough current.

Just a thought..

E3V3A, 2017-07-05:
Could it have something to do with the sample rate?
I wonder if this is also related to running in RTL-SDRs in a VM with the `PLL not locked` problem?

TLeconte, 2017-09-05:
in rtl.c try wit RTLMULT 160 and use last version (rtl init buff size fix) 
(sorry for the latency :-) )

ckuethe, 2018-01-09:
This can probably be closed now. I've been running acarsdec 3.5 for days, and have logged in excess of 100k messages.

closed 2018-01-11 by TLeconte