TLeconte/acarsdec #10: Cannot run 2 instances of acarsdec or 1x acarsdec and 1x dumpvdl2.

DickvanNoort, 2017-03-29:
When I run 2 instances of acarsdec (3.4) the second instance issues this error message:
       Failed to submit transfer 30!
       Read async -5
and aborts.

A similar error occurs when I run Dumpvdl2 and Acarsdec in that sequence.

After I changed rtl.c 
from
        r = rtlsdr_read_async(dev, in_callback, NULL, 32, RTLINBUFSZ);
to
        r = rtlsdr_read_async(dev, in_callback, NULL, 10, RTLINBUFSZ);
all works fine.

E3V3A, 2017-07-05:
Thanks, but to make this happen, please make a PR for @TLeconte to implement.

TLeconte, 2017-07-09:
OK. I see it too on a py 
I decreased to 8 buffers and it seems to work
I have commited the fix
thanks for for the bug report

DickvanNoort, 2017-07-09:
Thank you.

Cheers, 
Dick (PA2015)

closed 2017-08-05 by TLeconte