TLeconte/acarsdec #20: Wrong sample rate for RTL-SDR (R820T)

E3V3A, 2017-07-05:
Two problems here: 

1. Wrong sample rate. The RTL-SDR (820T2) max stable sample rate is: ` 2.4 MS/s` (USB 2.0)
2. There should be an automatic test for the provided frequencies to ensure that they are not separated more than 2 MHz. 

```
 # ./acarsdec -v -o 2 -p 1 -r 0 131.525 131.550 131.725 131.850 136.750 136.900

Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain : 49.600000
Frequencies too far apart
Set center freq. to -1Hz
[R82XX] PLL not locked!
Exact sample rate is: 2500000.107620 Hz
[R82XX] PLL not locked!
Decoding 6 channels
```

E3V3A, 2017-07-23:
### To fix issue 1.

To get correct sample rates replace [this](https://github.com/TLeconte/acarsdec/blob/969ec3c6ea4de5dba3e1e9b118f217779318b67b/rtl.c#L30-L31): 
```c
#define RTLMULT 200
#define RTLINRATE (INTRATE*RTLMULT)
```
with:
```c
//#define RTLMULT 200      // 2.5000 Ms/s
#define RTLMULT 192	// 2.4000 Ms/s
//#define RTLMULT 160	// 2.0000 Ms/s
//#define RTLMULT 144	// 1.8000 Ms/s
#define RTLINRATE (INTRATE*RTLMULT)
```

TLeconte, 2017-08-05:
Ok thanks
fix it by just returning 0 in ChooseFc (returning  -1 is not a good idea for an unsigned int return function)

closed 2017-08-05 by TLeconte

TLeconte, 2017-08-05:
ok will add theses comments in rtl.c and in README