TLeconte/acarsdec #21: Add automatic test for maximum provided frequency separation

E3V3A, 2017-07-05:
Since the RTL-SDR cannot fetch more than 2 MHz bandwidth, the frequencies provided in the list may not be separated by more than this. However, the binary happily accepts anything, without any warning with unknown consequences. 

I propose to implement an automatic check for this.

E3V3A, 2017-07-23:
Rewrite [this](https://github.com/TLeconte/acarsdec/blob/969ec3c6ea4de5dba3e1e9b118f217779318b67b/rtl.c#L136-L139):
```c
if ((Fd[nbch - 1] - Fd[0]) > RTLINRATE - 4 * INTRATE) {
		fprintf(stderr, "Frequencies too far apart\n");
		return -1;
}
```
To: 
```c
if ((Fd[nbch - 1] - Fd[0]) > RTLINRATE - 4 * INTRATE) {
		fprintf(stderr, "ERROR: Frequencies are too far apart: df=%.2f MHz\n", (float)(Fd[nbch - 1] - Fd[0]));
		exit(1);
}
```

PS. I compiled this in my head, so I haven't tested it...

TLeconte, 2017-08-05:
Ok thanks fix it by just returning 0 in ChooseFc (returning  -1 is not a good idea for an unsigned int return function)

closed 2017-08-05 by TLeconte