TLeconte/acarsdec #22: Request - Include Frequency in -N UDP output

acarslogger, 2017-07-07:
The UDP -N output allows the data to be sent to a remote machine. That machine may not know the acarsdec command line parameters used and will not know how to associate channel number to a specific frequency. Is it possible to change the UDP -N feed to include the frequency in place of the channel number e.g.
change
  (null) **2** 27/10/2016 08:56:52 0 -17 X .N223UA L _d 8 S59A UA0932
to
  (null) **131.725** 27/10/2016 08:56:52 0 -17 X .N223UA L _d 8 S59A UA0932

I'm not a C developer, so if anyone is willing to create a Pull Request I would appreciate it and hopefully it will be eventually included in the official released version of this excellent program.

Thanks
Stuart

ckuethe, 2018-01-09:
I'll do up a patch for this. I just posted a PR to also include frequency in JSON output

ckuethe, 2018-01-10:
Hm, this is going to break the wire protocol. It's easy enough to replace the channel number with the frequency in the outbound packet, but old `acarsserv`s won't know how to handle it. They're expecting a single character for the channel.

I suppose I can try include the frequency on the end (` 131.725MHz`) and make acarsserv attempt to find "%3.3fMHz" at the end of the message. And then if it finds that, overwrite the channel with frequency in memory before writing it to the database so that old clients will log channels 1-8, new clients will log 131725000 (Hz), and the database schema won't need to change.

Is it time to start versioning the protocol and the database?

ckuethe, 2018-01-11:
@TLeconte how would you feel about using JSON as new protocol wire format, now that #23 has been merged? There are a few pretty small json parsers out there such as [cjson](https://github.com/DaveGamble/cJSON) and [jsmn](https://github.com/zserge/jsmn).

For backwards compatibility I'm thinking of adding a new logging UDP logging style (`-J`?) and then people can do whatever they want with the logs, eg. a script that logs to mongodb, mysql, or a new acarsserv that understands JSON input?

TLeconte, 2018-01-11:
yes it's a possibility.
It will be great to do the same for vdlm2dec  ....
acarsserv could be left untouched and slowly obsoleted if you keep the -N option.
I know that acarsdec is used by lot of people , but I don't know for acarsserv ...

acarslogger, 2018-01-12:
@ckuethe & @TLeconte 
Thanks for your continued interest in this request. I thought it would be a simple change!
Stuart

ckuethe, 2018-01-17:
@acarslogger please try the patchset in #31

acarslogger, 2018-01-17:
@ckuethe 
Hi Chris

I've tested the -j output and it appears to be in JSON format (not that I'm an expert!) and does include the channel number and associated frequency. Thanks.

It appears that you can have either the -N or the -j option, not both. Both can be defined on the acarsdec command line, but only the first UDP command listed will be enabled. Is this intentional?

As the -N output format is not going to be changed to include the frequency, I need to change my little application I've written which parses the -N output so it can now read the -j JSON format. Another challenge / learning curve!

Maybe a dumb Github question --- how was I supposed to test the Patch? I just copied and pasted the modified files to my Raspberry Pi. Only took a few minutes. Looking at the help for Github is talks about the owner of the repository being able to automatically download the patch for testing. Is it possible as a user to download the patch to my Raspberry Pi with some command like 'git pull Patch_No' ?

Thanks for your help.
Stuart

ckuethe, 2018-01-17:
If you add `.patch` to a github change url, you get a raw diff. So you could do `wget https://github.com/TLeconte/acarsdec/pull/31.patch` on your Pi, which would give you the aggregation of all my changes in #31 . Then you could apply those changes with the `patch` command.

If your app is in python, it's just about the easiest thing in the world to ingest JSON into a dict: `parsed = json.loads(something)`. There are similar JSON parsers for other languages.

For now, it is intentional that only one of `-N` or `-j` is used. No efforts were made to have them both be active at the same time, because that might do weird things to the database. I should probably enforce some mutual exclusion.

acarslogger, 2018-01-17:
@ckuethe 
Thanks for my dummies Github commands. Worked OK.

Restricting to 1 feed (either -N or -j) may not be the best long term solution.
It's possible that a user may want to feed an acarsserv as well as another application that accepts an acarsdec -N and/or -j UDP feed. With a restriction of only one -N or -j feed option this wouldn't be permitted.
It maybe tricky, but I think you should allow for multiple -j / -N UDP feeds.
Any thoughts?
Stuart

TLeconte, 2018-01-17:
I fact, @ckuethe have done a such great job, that acarsserv accepts json format too now .
If your application is in C, read the @ckuethe changes to acarsserv to see how to easly parse the json.
More modern languages have often a more integrated  json parser.

ckuethe, 2018-01-17:
I thought about having multiple network logging targets, but that would require some re-architecting:

* acarsdec would need parse multiple logging options
* acarsdec would need to manage lists of logging targets
* acarsserv would need to understand multiple input sockets
* the database would need to have some deduplication or uniqueness indexes on the messages

There is a concept of "none, one, or infinity": don't allow/support something, allow a single exception, or design something arbitrarily scalable. Ten, hundred, thousand, million, billion, trillion... it shouldn't matter until you run into a hardware limit. In this case, I'm going to punt and go with a single log target or a single input protocol. I'm not opposed to having multiple targets and might look into it when get through some of my other projects;if you want to try your hand at implementing multi-logging before I do, I'll review your diffs.

acarslogger, 2018-01-17:
I started this issue/request/discussion as an aviation enthusiast, wanting to enhance acarsdec to provide the frequency instead of a channel number. It seems to be getting more and more complicated!

An idea.
acarsdec provides a dedicated feed (-n) to PlanePlotter in PP format.
Why not use one of the other feeds (e.g. -N) to feed dedicated data to acarsserv. It could have some specific data element/format that acarsserv will decode. If it's not in the correct format it throws the UDP packet away.
How does acarsserv cope with a Planeplotter feed '-n' that is sent to it? Does it process it or throw it away?

So ...
Have a '-n' feed in PP format.
Have a '-N' feed in acarsserv format. This could be in plain text or JSON format.
Have a '-XYZ' feed in either plain text or JSON format that can be used to send acars data to 1 or more other UDP destinations?

I expect there are intricacies and backward compatibility issues that prevent this simplistic approach from being implemented!

Stuart

ckuethe, 2018-01-17:
`-N` should not be "plain text or JSON format." It must remain in the plain text ("classic") format because the logging channel is one-way. An acarsserv process has no way to signal the client to request a particular format. Additionally, the protocol has no version information and there is no way to signal that the semantics of a message might have changed. The classic protocol is frozen in time forever.
https://github.com/TLeconte/acarsdec/blob/master/output.c#L143

I could make acarsserv attempt to parse a message as both JSON and classic format, taking which ever one succeeded, but the parser looks kind of fragile and I didn't want to touch it any more than necessary. 
https://github.com/TLeconte/acarsdec/blob/master/acarsserv.c#L300

Perhaps acarsserv development stop and a new daemon would be written. Python would probably be a good language due to its portability, robust JSON library, regex support, good database abstractions (so you could target sqlite, postgres, mysql, mongodb, mssql, oracle, etc.), event-driven networking frameworks, ... many features that would make this logging daemon simple to write and maintain.

acarslogger, 2018-01-18:
I'm a bit confused.

> For now, it is intentional that only one of -N or -j is used. No efforts were made to have them both be active at the same time, because that might do weird things to the database. I should probably enforce some mutual exclusion.

But acarsdec allows both a -n and a -N feed. How does acarsserv prevent weird thing happening if both the -n and -N point to acarsserv?

Why should the -N and the -j UDP be exclusive?

The '-n' UDP feed is in Planeplotter format.
The '-N' UDP feed is in plain text ("classic") format for acarsserv.
The '-j' UDP feed is in JSON format for transmitting to any other application.

I would hope that if acarsserv receives a UDP packet that is not in the 'plain text ("classic") format' it throws it away.

Why restrict acarsdec to only 1 -N or -j UDP feed?

closed 2018-01-20 by TLeconte

ckuethe, 2018-02-01:
@acarslogger I restricted it to one because at the time I didn't want to implement the other logic necessary to track which server gets Planeplotter format, which server gets classic format, and which one gets JSON. 

As best I can tell Acarsserv ignores planeplotter format; more correctly, it drops anything that isn't [properly formatted classic](https://github.com/TLeconte/acarsdec/blob/4ec2928f5ed046ea7bb5431847cd49a08bf53305/acarsserv.c#L297) protocol.

If you look at the the code [before JSON mode](https://github.com/TLeconte/acarsdec/blob/4ec2928f5ed046ea7bb5431847cd49a08bf53305/acarsdec.c#L168) you'll see that `netout` controlled the output format, and `Rawaddr` was the log target and the last `-n` or `-N` would win. With `-J` that behavior is embraced and extended.

Eventually I may be motivated to separate the log targets for independent processing in the output loop.

TLeconte, 2018-02-01:
note that I change the parameters of acarsserv and vdlm2dec recently to have a more coherent system, now   :  

- acarssdec  send in json with -j addr:port , planeplotter -n addr:port and its own format with -N addr:port 
- acarsserv receive in json with -j addr:port and  in acarsdec own format with -N addr:port 
-  vdlm2dec send in json format only with -j addr:port 

hope it helps

PR #31 (2018-01-20): add "-j" flag to log over the network in json mode
This diff to adds network logging in JSON mode. In order to simplify the JSON generation and parsing, I've included a copy of [cJSON](https://github.com/DaveGamble/cJSON) - this takes care of all the escaping and stuff.

This should resolve #22 without breaking any existing uses. When JSON is ingested into acarsserv, the channel is no longer the configured channel from 1-8. Instead, it is the received frequency in Hz, eg. `131550000`.