TLeconte/acarsdec #24: Document the error correction algorithm

dev-zzo, 2018-01-09:
Hello Thierry,

First let me thank you for this great piece of software. It does an amazing job indeed.

I would like to request you to please document the exact algorithm you have implemented to correct errors. I have figured out most of it but the syndrome part still throws me off as I am not super good with CRC maths.

My final goal is to improve the algorithm to ensure the corrected messages are well formed e.g. the aircraft reg. number doesn't contain oddball characters. Ideally I'd appreciate a callback function inserted into the error correction code which I could then modify. I'd also need to ensure the algorithm actually completes. :-)

sv1, 2018-01-15:
Hello Thierry,

I do not know if it has anything to do with what dev-zzo says, but since the commits on Jan 11, I can not decode any message, anymore.  All I get is too many parity errors.

dev-zzo, 2018-01-15:
@sv1 It is not related; I were asking for something that was actually working. :-) Please create a new issue and if possible try to spot the exact offending commit.

TLeconte, 2018-01-15:
Yes open a new issue.
I just recheck now with my  airspy and it seems  working ....
Hope I have not break rtl as I don't have one any more now.
Could someone else confirm that it works (or not)  with rtl  ?

TLeconte, 2018-01-25:
@sv1 I just fixed a nasty cast problem introduced by commit [edd0bd70b1bac1d](https://github.com/TLeconte/acarsdec/commit/058eb01964d655ad32f33ff83edd0bd70b1bac1d)
It was working on ARM but not on x86_64  :-(
Could you try last version ?

sv1, 2018-01-26:
> 5493e4bb","name":"GitHub"},"entity":{"external_key":"github/TLeconte/
> acarsdec","title":"TLeconte/acarsdec","subtitle":"GitHub
> repository","main_image_url":"https://cloud.githubusercontent.com/ass
> ets/143418/17495839/a5054eac-5d88-11e6-95fc-
> 7290892c7bb5.png","avatar_image_url":"https://cloud.githubusercontent
> .com/assets/143418/15842166/7c72db34-2c0b-11e6-9aed-
> b52498112777.png","action":{"name":"Open in
> GitHub","url":"https://github.com/TLeconte/acarsdec"}},"updates":{"sn
> ippets":[{"icon":"PERSON","message":"@TLeconte in #24: @sv1 I just
> fixed a nasty cast problem introduced by commit
> [edd0bd70b1bac1d](https://github.com/TLeconte/acarsdec/commit/058eb01
> 964d655ad32f33ff83edd0bd70b1bac1d)\r\nIt was working on ARM but not
> on x86_64 :-(\r\nCould you try last version
> ?"}],"action":{"name":"View
> Issue","url":"https://github.com/TLeconte/acarsdec/issues/24#issuecom
> ment-360397996"}}}

> @sv1 I just fixed a nasty cast problem introduced by commit
> edd0bd70b1bac1d
> It was working on ARM but not on x86_64 :-(
> Could you try last version ?

I am on a trip now Thierry, as soon as I get back I will do.

thank you very much.

sv1, 2018-03-10:
Sorry for the delay!!!

Everything is working great

Thanks Thierry

rfrht, 2018-08-19:
@TLeconte bug keeping: You may now close this bug report! ;-)

TLeconte, 2018-08-19:
ok

closed 2018-08-19 by TLeconte