TLeconte/acarsdec #29: Receiver goes deaf after some time

dev-zzo, 2018-01-16:
So this is an issue I have seen quite often on my setup: the receiver goes deaf after running for random period of time, no messages are decoded even though the signal strength has not changed. It seems hard to provide a reproducer though. :-/

After some staring at the code I seem to have managed to come up with a patch that, when applied, appears to mitigate the issue; at least, my systems now do not exhibit deafness any more.

https://github.com/dev-zzo/acarsdec/commit/dfbd30bcf2e4cd8bb3b87ccfeabe662b82a5842b

Please let me know if I should make a pull request for you to integrate the changes.

ckuethe, 2018-01-16:
I would argue that it's always worthwhile to make a PR, even for changes that may not be immediately integrated.

Anyhoo, I haven't seen this issue but I'll run your patch on my system to regression test it.

ckuethe, 2018-01-17:
You should do up a PR. your patch doesn't break my system after about 28hrs of operation

dev-zzo, 2018-01-18:
Created PR#32.

TLeconte, 2018-01-20:
So, sometime complexity is not better.
I decided to switch from a 2nd order PLL to 1st order one. It does not change the result of my tests ...

No more Mska, no more problems.

@dev-zzo could you try y last commit ?

dev-zzo, 2018-01-21:
Would this change impact BER in a negative way?

TLeconte, 2018-01-21:
Don't think so. There is theory and practice...
I have some test files and I don't get less  decoded messages with this new version.

closed 2018-01-21 by TLeconte