TLeconte/acarsdec #41: JSON format issue

rfrht, 2018-08-19:
Hello there, Thierry!

First of all, thank you VERY much for acarsdec!

I am trying to use the `acarsdec` output and parse it with python module `json.tool`. However, it is not accepted at all: It always fail with a formatting complaint.

Let the snippet that you have provided in readme.md:

````
{"timestamp":1516206744.1849549,"channel":2,"freq":130.025,"level":-22,"error":0,"mode":"2","label":"H1","block_id":"6","ack":false,"tail":".N842UA","flight":"UA1412","msgno":"D04G","text":"#DFB9102,0043,188/9S101,0039,181/S0101,0043,188/0S100,0039,182/T1100,0043,188/1T099,0039,182/T2099,0043,189/2T098,0039,182/T3098,0043,189/3T097,0039,182/T4098,0043,189/4T097,0039,183/T5098,0043,189/5T097,0039,1","end":true,"station_id":"sigint"}
{"timestamp":1516206745.249615,"channel":2,"freq":130.025,"level":-24,"error":2,"mode":"2","label":"RA","block_id":"R","ack":false,"tail":".N842UA","flight":"","msgno":"","text":"QUHDQWDUA?1HOWGOZIT\r\n ** PART 01 OF 01 **\r\nHOWGOZIT 1412-17 SJC\r\nCI: 17        RLS: 01 \r\nSJC 1615/1625     171A\r\nBMRNG    1630 37  159-\r\nTIPRE    1638 37  145\r\nINSLO    1701 37  125\r\nGAROT    1726 37  106\r\nEKR      1800 ","end":true,"station_id":"sigint"}
{"timestamp":1516206747.0520389,"channel":2,"freq":130.025,"level":-24,"error":0,"mode":"2","label":"H1","block_id":"6","ack":"R","tail":".N842UA","flight":"UA1412","msgno":"D04G","text":"#DFB9102,0043,188/9S101,0039,181/S0101,0043,188/0S100,0039,182/T1100,0043,188/1T099,0039,182/T2099,0043,189/2T098,0039,182/T3098,0043,189/3T097,0039,182/T4098,0043,189/4T097,0039,183/T5098,0043,189/5T097,0039,1","end":true,"station_id":"sigint"}
{"timestamp":1516206752.622808,"channel":2,"freq":130.025,"level":-23,"error":0,"mode":"2","label":"RA","block_id":"S","ack":"6","tail":".N842UA","flight":"","msgno":"","text":"37   80\r\nDEN 1829/1837      67\r\n--SCHEDULED ARRIVAL--\r\n    1842/1850 (00.13E)\r\n** END OF PART 01 **\r\n","station_id":"sigint"}
````
And then saving it to a file and trying to parse it using `json.tool`:

    [root@radar acars]# python -mjson.tool tleconte.json 
    Extra data: line 2 column 1 - line 5 column 1 (char 436 - 1636)

Now, some data that I have saved here:

````
{"timestamp":1534695523.8791251,"channel":0,"freq":131.550,"level":-18,"error":0,"mode":"2","label":"RA","block_id":"L","ack":false,"tail":"PR-YRF","flight":"","msgno":"","text":"QUASRV1AD~1RA-1\r\nSA 19/16:00 RTD \r\nMETAR SBGR 191600Z 32003KT 9999 FEW030 24/14 Q1023=\r\nSA 19/16:00     \r\nSBRP 191600Z VRB03KT 9999 BKN035 28/09 Q1021=\r\nSA 19/16:00     \r\nSBKP 191600Z 35007KT 300V020 9999 SCT040 25/10 Q1","end":true,"station_id":"radar.vl"}
{"timestamp":1534695528.8761849,"channel":0,"freq":131.550,"level":-18,"error":0,"mode":"2","label":"RA","block_id":"M","ack":false,"tail":"PR-YRF","flight":"","msgno":"","text":"022=\r\n","station_id":"radar.vl"}
{"timestamp":1534696425.332144,"channel":0,"freq":131.550,"level":-18,"error":0,"mode":"2","label":"RA","block_id":"A","ack":false,"tail":"PR-ONC","flight":"","msgno":"","text":"QUSAOASO6~1LOADSHEET FINAL 1333 EDNO1\r\nO66020/19     19AUG18\r\nCGH SDU PR-ONC   2/4\r\nZFW 48707  MAX 54500 \r\nTOF 5300\r\nTOW 54007  MAX 64800 \r\nTIF 1816\r\nLAW 52191  MAX 57500   L\r\nUNDLD 5309\r\nPAX/105 TTL 106\r\nMACZFW   24.3\r\n","end":true,"station_id":"radar.vl"}
{"timestamp":1534696430.32501,"channel":0,"freq":131.550,"level":-17,"error":0,"mode":"2","label":"RA","block_id":"B","ack":false,"tail":"PR-ONC","flight":"","msgno":"","text":"MACTOW   24.1\r\nMACLAW   24.9\r\nSTAB:STANDARD           1.6 UP\r\nA105 \r\nSEATROW TRIM\r\nSI A318\r\nDOW 40260\r\nSERRA 679142\r\nPAX WEIGHTS USED A80 C40 I10\r\nPREPARED BY MEIRELES DUARTE/MICHAEL 55 \r\n11 990180136\r\nNOTOC: NO\r\nFUEL IN","end":true,"station_id":"radar.vl"}
{"timestamp":1534696435.322489,"channel":0,"freq":131.550,"level":-17,"error":0,"mode":"2","label":"RA","block_id":"C","ack":false,"tail":"PR-ONC","flight":"","msgno":"","text":" TANKS 5500\r\nLOAD IN CPTS 0/0 1/0 4/157 5/0 \r\nISSUANCE DATE AND TIME: 19AUG 16:33 UTC\r\nDOV LEANDRO 130026\r\n","station_id":"radar.vl"}
````

`acarsdec` CLI parameters:

    ./acarsdec -r 1 131.550 -o 4

Results:

    [root@radar acars]# python -mjson.tool ra.txt
    Extra data: line 2 column 1 - line 6 column 1 (char 449 - 1900)

Expected results. See the snippet:

Obtaining a Raw json content: 
````
[rfreire@tambot qpackage-1.0]$ curl -s --insecure https://api.iextrading.com/1.0/stock/AAPL/quote
{"symbol":"AAPL","companyName":"Apple Inc.","primaryExchange":"Nasdaq Global Select","sector":"Technology","calculationPrice":"close","open":213.47,"openTime":1534512600588,"close":217.58,"closeTime":1534536000445,"high":217.95,"low":213.16,"latestPrice":217.58,"latestSource":"Close","latestTime":"August 17, 2018","latestUpdate":1534536000445,"latestVolume":34998882,"iexRealtimePrice":null,"iexRealtimeSize":null,"iexLastUpdated":null,"delayedPrice":217.58,"delayedPriceTime":1534536000445,"extendedPrice":217.55,"extendedChange":-0.03,"extendedChangePercent":-0.00014,"extendedPriceTime":1534539553278,"previousClose":213.32,"change":4.26,"changePercent":0.01997,"iexMarketPercent":null,"iexVolume":null,"avgTotalVolume":25095709,"iexBidPrice":null,"iexBidSize":null,"iexAskPrice":null,"iexAskSize":null,"marketCap":1050895299080,"peRatio":21,"week52High":217.95,"week52Low":149.16,"ytdChange":0.29756620725736654}[rfreire@tambot qpackage-1.0]$ 
````

And then parsing it with `json.tool`:

````
[rfreire@tambot qpackage-1.0]$ curl -s --insecure https://api.iextrading.com/1.0/stock/AAPL/quote | python -mjson.tool
{
    "avgTotalVolume": 25095709, 
    "calculationPrice": "close", 
    "change": 4.2599999999999998, 
    "changePercent": 0.019970000000000002, 
    "close": 217.58000000000001, 
    "closeTime": 1534536000445, 
    "companyName": "Apple Inc.", 
    "delayedPrice": 217.58000000000001, 
    "delayedPriceTime": 1534536000445, 
    "extendedChange": -0.029999999999999999, 
    "extendedChangePercent": -0.00013999999999999999, 
    "extendedPrice": 217.55000000000001, 
    "extendedPriceTime": 1534539553278, 
    "high": 217.94999999999999, 
    "iexAskPrice": null, 
    "iexAskSize": null, 
    "iexBidPrice": null, 
    "iexBidSize": null, 
    "iexLastUpdated": null, 
    "iexMarketPercent": null, 
    "iexRealtimePrice": null, 
    "iexRealtimeSize": null, 
    "iexVolume": null, 
    "latestPrice": 217.58000000000001, 
    "latestSource": "Close", 
    "latestTime": "August 17, 2018", 
    "latestUpdate": 1534536000445, 
    "latestVolume": 34998882, 
    "low": 213.16, 
    "marketCap": 1050895299080, 
    "open": 213.47, 
    "openTime": 1534512600588, 
    "peRatio": 21, 
    "previousClose": 213.31999999999999, 
    "primaryExchange": "Nasdaq Global Select", 
    "sector": "Technology", 
    "symbol": "AAPL", 
    "week52High": 217.94999999999999, 
    "week52Low": 149.16, 
    "ytdChange": 0.29756620725736654
}
````

Can you please check it?

I appreciate it a lot.

TLeconte, 2018-08-19:
I did not write the json output. 
I don't see anything wrong with the acarsdec output but I'm not a json expert.
Perhaps it's just because your parser want a comma at each line end.
In fact, acarsdec json output is more a log off json formated lines, not really a json file.
Could you try the two acarsdec output examples  and add a coma at each line  end ?

ckuethe, 2019-03-18:
When I created the JSON output, I intended it to be very similar to the other line oriented protocol where each line stands alone, independent of all others. This means I don't risk having to maintain potentially huge amounts of state when reading in a large file of observations...

closed 2019-06-22 by TLeconte