TLeconte/acarsdec #51: libacars UDP feed

acarslogger, 2018-12-01:
Hi Thierry,

The current -j UDP output generates JSON output of the basic acars messages as well as the additional 'acarsdec' decoded data of departure and destination - JSON "depa" and "dsta".

With the adiition of libacars would it be possible to include the additional 'libacars' decoded data in the UDP output; maybe a 'libacars' JSON field with the data.

Stuart

TLeconte, 2018-12-03:
Good idea,, but ARIC 622 decoding (libacars)  is complex and man will need lots of fields.
Perhaps position / altitude / speed could be a good start.
By the way, my vdlm2dec could now use libacars too.

acarslogger, 2018-12-03:
Hi Thierry,

Sorry, I wasn't suggesting trying to parse all the individual data elements into individual JSON fields.
Put all the libacars generated text output into 1 JSON field. That way it's up to the downstream system that receives the UDP packet to use it as it wants. No complex parsing of individual pieces of data for you!

Stuart
p.s. Good to know about vdlm2dec -- GREAT - that was going to be my next issue!!!

szpajder, 2018-12-03:
If it's of interest to you, JSON output generator is a TODO feature targeted for a next libacars release.

But first, I'd like to make the actual 1.0 release, as it's not there yet ;)

jmceara, 2019-03-15:
> Hi Thierry,
>
> Sorry, I wasn't suggesting trying to parse all the individual data elements into individual JSON fields.
> Put all the libacars generated text output into 1 JSON field. That way it's up [...]

That would be great if client could send decoded msg to server, even in one field!

szpajder, 2019-03-15:
Don't you think this is better?

```
$ ./decode_arinc d /CCUCAYA.ADS.9V-SKA072621E05B8309C41A741D1725D95842F349C404B00C4C9073CB28201176CD610E5D88EB40000F5E49A280001018379E700D261CC8596709C40069260540509589C4003344 2>/dev/null|tail -1 |json_pp 
{
   "arinc622" : {
      "msg_type" : "adsc_msg",
      "air_addr" : ".9V-SKA",
      "adsc" : {
         "err" : false,
         "tags" : [
            {
               "basic_report" : {
                  "lon" : 16.086044,
                  "pos_accuracy_nm" : 0.25,
                  "nav_redundancy" : true,
                  "lat" : 53.623581,
                  "ts_sec" : 1693,
                  "tcas_avail" : true,
                  "alt" : 40000
               }
            },
            {
               "fixed_projection" : {
                  "alt" : 40000,
                  "lon" : 11.768589,
                  "eta_sec" : 1200,
                  "lat" : 53.225155
               }
            },
            {
               "flight_id" : {
                  "flight_id" : "SIA322  "
               }
            },
            {
               "earth_ref_data" : {
                  "true_trk_valid" : true,
                  "vspd_ftmin" : 0,
                  "gnd_spd_kts" : 470.5,
                  "true_trk_deg" : 263.056641
               }
            },
            {
               "air_ref_data" : {
                  "true_hdg_deg" : 265.166016,
                  "vspd_ftmin" : 0,
                  "true_hdg_valid" : true,
                  "spd_mach" : 0.837
               }
            },
            {
               "meteo_data" : {
                  "wind_spd_kts" : 24,
                  "temp_c" : -50,
                  "wind_dir_true_deg" : 312.1875,
                  "wind_dir_valid" : true
               }
            },
            {
               "predicted_route" : {
                  "next_next_wpt" : {
                     "lat" : 53.466339,
                     "lon" : 14.165154,
                     "alt" : 40000
                  },
                  "next_wpt" : {
                     "alt" : 40000,
                     "lon" : 15.715256,
                     "eta_sec" : 105,
                     "lat" : 53.5956
                  }
               }
            }
         ]
      },
      "crc_ok" : true,
      "gs_addr" : "CCUCAYA"
   }
}
```

closed 2019-06-22 by TLeconte