TLeconte/acarsdec #58: Doesn't work with AirSpy-mini

jmceara, 2019-03-15:
Tried latest version (from git) using latest version of libairspy, no success.

```
pi@raspberrypi:~/acarsdec/build $ airspy_info
airspy_lib_version: 1.0.9

Found AirSpy board 1
Board ID Number: 0 (AIRSPY)
Firmware Version: AirSpy MINI v1.0.0-rc9-0-ga56adfd 2016-06-12
Part ID Number: 0x6906002B 0x00000030
Serial Number: 0x04A464C83488940B
Supported sample rates:
        6.000000 MSPS
        3.000000 MSPS
Close board 1
pi@raspberrypi:~/acarsdec/build $ ./acarsdec -v -s 131.523
did not find needed sampling rate
Unable to init input

```

kevinelliott, 2020-07-31:
Definitely want Airspy Mini support. Any chance we can have this @TLeconte ?

guidocioni, 2022-04-10:
To add support for Airspy Mini you can manually modify the Sample Rate by setting `AIRMULT = 480` here

https://github.com/TLeconte/acarsdec/blob/092989e389bfcdbd8bca0b94608f87b709c4a8e4/air.c#L31

and compile again.
This will force a sample rate of 6 MS instead than 5 MS

TLeconte, 2022-06-11:
recent fixes seems to work with airspy mini

closed 2022-06-11 by TLeconte