TLeconte/acarsdec #63: rtlsdr - SDR:Device not found

Twilight-Logic, 2019-04-28:
I am getting the following:
I am using this imple command line:

`cmd /c acarsdeco2.exe --device-serial 1 --freq-correction 0 --freq 131725000 --freq 131525000 --freq 131814000 --http-port 8082 --vrs-url http://127.0.0.1:83
`

I am getting this response:

> AcarSDeco2 v.20181201
> Using device:  0 mnf: Realtek prod: RTL2838UHIDIR serial: 00000001
> Sampling rate: 1400000.018544 sps
> SDR Device Not found
> Press any key to continue . . .

Evidently identifies the dongle and then says Device not found? Uh?

acarslogger, 2019-04-30:
This github is for acars decoding using **acarsdec**.
You appear to be using the Windows acarsdeco2 application.
Maybe look at https://radarspotting.com/forum/index.php/board,57.0.html?PHPSESSID=91bd7403abafa6bcc5cfaec4f5e8aa59 .

Stuart

Twilight-Logic, 2019-05-11:
Thanks for your reply and please forgive my confusion. It was my impression that acarsdec was cross-platform and acarsdec2 was the windows version. I would actually prefer to use this on Linux, particularly on Raspian so actually this is the version I perhaps need.

Since I have both an rtlsdr and an sdrplay, is it possible to compile with support for both?

If I try specifying both, i.e -Drtl=ON -Dsdrplay=ON, I get and error indicating duplicate variables:

```
$ make
[ 11%] Building C object CMakeFiles/acarsdec.dir/acars.c.o
In file included from /home/johnc/src/acarsdec/acars.c:20:0:
/home/johnc/src/acarsdec/acarsdec.h:62:8: error: duplicate member ‘Fr’
  float Fr;
        ^~

```
I guess I can compile separate versions for each radio if required.

closed 2019-06-22 by TLeconte