TLeconte/acarsdec #69: Cannot monitor multiple frequencies

tinjaw, 2020-07-28:
This line will produce output:

acarsdec -v -o 4 -g 500 -i TJ-MCI1-ACARS -j feed.acars.io:5550 -r 111 130.025

This line will NOT produce output:

acarsdec -v -o 4 -g 500 -i TJ-MCI1-ACARS -j feed.acars.io:5550 -r 111 130.025 130.450 130.825 131.125 131.425 131.550 131.725 131.850

n.b. 130.025 is included in both commands.

tinjaw, 2020-07-28:
Here is a better example. This is over a 10-15 minute period.

```
pi@acars:~ $ acarsdec -v -o 4 -g 500 -i TJ-MCI1-ACARS -j feed.acars.io:5550  -r 0 130.025
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 333

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain : 49.600000
Set center freq. to 130050000Hz
Exact sample rate is: 2000000.052982 Hz
Decoding 1 channels
Allocating 4 zero-copy buffers
{"timestamp":1595948016.1621971,"station_id":"TJ-MCI1-ACARS","channel":0,"freq":130.025,"level":-16,"error":0,"mode":"2","label":"_d","block_id":"4","ack":false,"tail":"N210UA","flight":"UA2465","msgno":"S70A"}
{"timestamp":1595948040.818115,"station_id":"TJ-MCI1-ACARS","channel":0,"freq":130.025,"level":-16,"error":0,"mode":"2","label":"H1","block_id":"5","ack":false,"tail":"N210UA","flight":"UA2465","msgno":"D62A","text":"E24C052102465KORDKDEN0728201453ER 89\r\n 37999-2052638306426960  18  44XXXX32355\r\n777106A005030  870  60   9   4   41\r\n777022B00  881  54  11   5   30\r\n11761176 8364019 850 6824 6700565\r\n11761176 8414080 854 ","end":true}
^Creceive signal 2 exiting
pi@acars:~ $ acarsdec -v -o 4 -g 500 -i TJ-MCI1-ACARS -j feed.acars.io:5550  -r 0 130.025 130.450 130.825 131.125 131.425 131.550 131.725 131.850
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 333

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain : 49.600000
Set center freq. to 131000000Hz
Exact sample rate is: 2000000.052982 Hz
Decoding 8 channels
Allocating 4 zero-copy buffers
#1 too many parity errors
#8 too many parity errors
#1 too many parity errors
#1 too many parity errors
#1 too many parity errors
#1 too many parity errors
#1 too many parity errors
#7 too many parity errors
#1 too many parity errors
^Creceive signal 2 exiting
pi@acars:~ $ acarsdec -v -o 4 -g 500 -i TJ-MCI1-ACARS -j feed.acars.io:5550  -r 0 130.025
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 333

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain : 49.600000
Set center freq. to 130050000Hz
Exact sample rate is: 2000000.052982 Hz
Decoding 1 channels
Allocating 4 zero-copy buffers
{"timestamp":1595948221.0786741,"station_id":"TJ-MCI1-ACARS","channel":0,"freq":130.025,"level":-10,"error":0,"mode":"2","label":"H1","block_id":"2","ack":false,"tail":"N468UA","flight":"UA2420","msgno":"D07A","text":"A38/A32038,1,1/C1TRP,145639,409934,-0934763,380,01345/C2459,0001878,0018340,0580,29051/C300988,00968,00005,-0003,-49,00,N0,01/C4249,8,0001848/"}
#1 parity error(s): 1
#1 crc error
#1 errors fixed
{"timestamp":1595948222.6353021,"station_id":"TJ-MCI1-ACARS","channel":0,"freq":130.025,"level":-20,"error":1,"mode":"2","label":"Q0","block_id":"9","ack":false,"tail":"N879UA","flight":"UA0348","msgno":"S81A"}
{"timestamp":1595948227.636785,"station_id":"TJ-MCI1-ACARS","channel":0,"freq":130.025,"level":-21,"error":0,"mode":"2","label":"_d","block_id":"Z","ack":"9","tail":"N879UA"}
{"timestamp":1595948228.6150889,"station_id":"TJ-MCI1-ACARS","channel":0,"freq":130.025,"level":-17,"error":0,"mode":"2","label":"_d","block_id":"1","ack":"E","tail":"N686FE","flight":"FX0677","msgno":"S68A"}
^Creceive signal 2 exiting
pi@acars:~ $ 

```

tinjaw, 2020-07-28:
Someone is a chat forum suggested that it might be locking on to one freq that has lots of noice.

Is there a way to see in real time what freqs are being scanned?

tinjaw, 2020-07-28:
In light of the above suggestion, I tried only two freqs that showed messages, and it worked. Further suggesting that it is some noisy channel grabbing all the attention.

```
pi@acars:~ $ acarsdec -v -o 4 -g 400 -i TJ-MCI1-ACARS -j feed.acars.io:5550  -r 0 131.550 130.025
Found 2 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 111
  1:  Realtek, RTL2838UHIDIR, SN: 222

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain : 40.200000
Set center freq. to 131000000Hz
Exact sample rate is: 2000000.052982 Hz
Decoding 2 channels
Allocating 4 zero-copy buffers
{"timestamp":1595952061.392236,"station_id":"TJ-MCI1-ACARS","channel":0,"freq":131.550,"level":-23,"error":0,"mode":"2","label":"SQ","text":"02XAMCIKMCI13917N09443WV136975/ARINC"}
#2 too many parity errors
{"timestamp":1595952086.614203,"station_id":"TJ-MCI1-ACARS","channel":1,"freq":130.025,"level":-25,"error":0,"mode":"2","label":"Q0","block_id":"3","ack":false,"tail":"N659FE","flight":"FX3649","msgno":"S70A"}
{"timestamp":1595952102.753336,"station_id":"TJ-MCI1-ACARS","channel":1,"freq":130.025,"level":-26,"error":0,"mode":"2","label":"Q0","block_id":"3","ack":false,"tail":"N659FE","flight":"FX3649","msgno":"S70A"}
{"timestamp":1595952104.473505,"station_id":"TJ-MCI1-ACARS","channel":1,"freq":130.025,"level":-23,"error":0,"mode":"2","label":"H1","block_id":"8","ack":false,"tail":"N926UW","flight":"AA1929","msgno":"D22A","text":"A38/A32138,1,1/C1TRP,160137,KPDX,KCLT,08,8,83602/C2415192,-0961178,350,01633,458,0230,1/C3413190,-0956243,349,01630,457,0230,1/C4411161,-0951326,350,01627,463,0229,1/C5409125,-0946484,349,01624,457,0230,1/C","end":true}
{"timestamp":1595952110.9455781,"station_id":"TJ-MCI1-ACARS","channel":1,"freq":130.025,"level":-22,"error":0,"mode":"2","label":"H1","block_id":"9","ack":false,"tail":"N926UW","flight":"AA1929","msgno":"D22B","text":"6407064,-0941665,350,01621,459,0230,1/C70003128,0000798,0497,30371,17879,-44/C80000199,0000053,0486,30026,17599,-44/C90000302,0000112,0445,30449,17279,-43/C00000087,0000059,0445,30254,16999,-43/E10000035,00","end":true}
{"timestamp":1595952116.67995,"station_id":"TJ-MCI1-ACARS","channel":1,"freq":130.025,"level":-21,"error":0,"mode":"2","label":"H1","block_id":"0","ack":false,"tail":"N926UW","flight":"AA1929","msgno":"D22C","text":"A38/A32138,1,1/C1TRP,160137,KPDX,KCLT,08,8,83602/C2415192,-0961178,350,01633,458,0230,1/C3413190,-0956243,349,01630,457,0230,1/C4411161,-0951326,350,01627,463,0229,1/C5409125,-0946484,349,01624,457,0230,1/C6407064,-0941665,350,01621,459,0230,1/C70003128,0000798,0497,30371,17879,-44/C80000199,0000053,0486,30026,17599,-44/C90000302,0000112,0445,30449,17279,-43/C00000087,0000059,0445,30254,16999,-43/E10000035,0000014,0474,29813,16679,-43/E20022258,00998,00978,00004,-0012,00,264/E3000,01/"}
{"timestamp":1595952173.1653471,"station_id":"TJ-MCI1-ACARS","channel":0,"freq":131.550,"level":-23,"error":0,"mode":"2","label":"SQ","text":"02XAMCIKMCI13917N09443WV136975/ARINC"}
{"timestamp":1595952180.824697,"station_id":"TJ-MCI1-ACARS","channel":1,"freq":130.025,"level":-20,"error":0,"mode":"2","label":"H1","block_id":"1","ack":false,"tail":"N926UW","flight":"AA1929","msgno":"F27A","text":"POSN40357W093578,LMN,160257,350,STL,162549,PXV,M44,29547,16508A0"}
{"timestamp":1595952187.9522851,"station_id":"TJ-MCI1-ACARS","channel":1,"freq":130.025,"level":-20,"error":0,"mode":"2","label":"_d","block_id":"2","ack":"Y","tail":"N926UW","flight":"AA1929","msgno":"S45A"}
{"timestamp":1595952188.5252981,"station_id":"TJ-MCI1-ACARS","channel":1,"freq":130.025,"level":-20,"error":0,"mode":"2","label":"H1","block_id":"3","ack":false,"tail":"N926UW","flight":"AA1929","msgno":"F28A","text":"RESREQ/AK,1158AF6"}

```

TLeconte, 2020-07-28:
acarsdec don't scan, it receives _simultaneously_ all the frequencies and, no, it does not matter if a frequency is noisy.
But  the more frequencies you use , the more CPU power you need and I fear you are at the limit of your CPU.
Try the command top to see how much CPU acarsdec use , and vary the number of frequencies to see the variation in CPU

tinjaw, 2020-07-28:
You are spot on. Two Freqs and I am at 100%. Just a raspberry pi 3B+.

Thanks

closed 2020-07-28 by tinjaw