TLeconte/acarsdec #76: Multiple concurrent socket output.

DickvanNoort, 2021-03-08:
Besides the output to a logfile only one of the 3 (n, N and J) sockets outputs seems to be supported.
If both -n and -j are selected only one output seems to be generated. No error message is shown.
If only -n or -j is selected it works fine.  
The switch code in acarsdec.c  seems to fill 'netout' with only the last of the 3 possible parameters that appear in the parameter list.  

My request is to support multiple concurrent socket output feeds so I can feed the data in both PP and in JSON format.

Cheers,
Dick

jaymot123, 2022-05-03:
I would like to be able to send messages in JSON format to two different ACARS aggregator sites, so I'd like to request that acarsdec be able to support multiple -j options. Basically that acarsdec not be limited to only one -j, -n and/or -N output. Thank you.

DickvanNoort, 2022-05-03:
> I would like to be able to send messages in JSON format to two different ACARS aggregator sites, so I'd like to request that acarsdec be able to support multiple -j options. Basically that acarsdec not be limited to only one -j, -n and/or -N output. 

 

If the format you want to send to both destinations is the same (like both JSON) you can use an UDP repeater like this one:

https://github.com/UnionPacific/udp-repeater  

 

Only if the format is not the same the program has to be changed.

 

Cheers,
Dick

jaymot123, 2022-05-04:
Thank you, DickvanNoort. That should do exactly what I'll want when the time comes to feed the second ACARS aggregation site.