TLeconte/acarsdec #77: Program not launching

gryzel1, 2021-04-22:
Hello,
I just installed the current version of acarsdec from this repository, and can't find a way to make it work.
Actually, regardless what parameters I choose, the program only returns the help menu.

I am using an RTL-SDR Antenna with needed libraries installed, on Debian 10 Buster (x64 intel).
Can you help me on this ? Thanks a lot

Here's what is happening:
```
(base) mathis@debian:~/Linux-version$ acarsdec -o1 -v -p -22 -g 400 -r 0 131.725
Acarsdec/acarsserv 3.5 Copyright (c) 2017 Thierry Leconte
(libacars 2.0.0)

Usage: acarsdec  [-v] [-o lv] [-t time] [-A] [-n ipaddr:port] [-l logfile [-H|-D]]

 -v			: verbose
 -A			: don't display uplink messages (ie : only aircraft messages)

 -o lv			: output format : 0 : no log, 1 : one line by msg, 2 : full (default) , 3 : monitor , 4 : msg JSON, 5 : route JSON

 -t time		: set forget time (TTL) in seconds for monitor mode (default=600s)
 -l logfile		: append log messages to logfile (Default : stdout).
 -H			: rotate log file once every hour
 -D			: rotate log file once every day
 -n ipaddr:port		: send acars messages to addr:port on UDP in planeplotter compatible format
 -N ipaddr:port		: send acars messages to addr:port on UDP in acarsdec native format
 -j ipaddr:port		: send acars messages to addr:port on UDP in acarsdec json format
 -i stationid		: station id used in acarsdec network format.
 -b filter		: filter output by label (ex: -b "H1:Q0" : only output messages  with label H1 or Q0)

Up to 8 channels may be simultaneously decoded
```

nsouleman, 2021-04-23:
Can't be reproduced here. Your command works for me.
Try to run your command using strace, you'll probably spot something relevant.

gryzel1, 2021-04-23:
Thank you for your help.
I tried using strace, and the answer looks like this:
```
execve("/usr/local/bin/acarsdec", ["acarsdec", "-o1", "-v", "-p", "-22", "-g", "400", "-r", "0", "131.725"], 0x7ffd0b3a0fc8 /* 49 vars */) = 0
brk(NULL)                               = 0x5588cf279000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (Aucun fichier ou dossier de ce type)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=119412, ...}) = 0
mmap(NULL, 119412, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7fe211a54000
close(3)                                = 0
openat(AT_FDCWD, "/usr/local/lib/libacars-2.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\20\246\2\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=788240, ...}) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fe211a52000
mmap(NULL, 626936, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fe2119b8000
mmap(0x7fe2119e2000, 180224, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2a000) = 0x7fe2119e2000
mmap(0x7fe211a0e000, 163840, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x56000) = 0x7fe211a0e000
mmap(0x7fe211a36000, 114688, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x7d000) = 0x7fe211a36000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libpthread.so.0", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0@l\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=146968, ...}) = 0
mmap(NULL, 132288, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fe211997000
mmap(0x7fe21199d000, 61440, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x6000) = 0x7fe21199d000
mmap(0x7fe2119ac000, 24576, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x15000) = 0x7fe2119ac000
mmap(0x7fe2119b2000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1a000) = 0x7fe2119b2000
mmap(0x7fe2119b4000, 13504, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7fe2119b4000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libm.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0p\322\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=1579448, ...}) = 0
mmap(NULL, 1581384, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fe211814000
mmap(0x7fe211821000, 651264, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xd000) = 0x7fe211821000
mmap(0x7fe2118c0000, 872448, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xac000) = 0x7fe2118c0000
mmap(0x7fe211995000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x180000) = 0x7fe211995000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\260A\2\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=1824496, ...}) = 0
mmap(NULL, 1837056, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fe211653000
mprotect(0x7fe211675000, 1658880, PROT_NONE) = 0
mmap(0x7fe211675000, 1343488, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x22000) = 0x7fe211675000
mmap(0x7fe2117bd000, 311296, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x16a000) = 0x7fe2117bd000
mmap(0x7fe21180a000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1b6000) = 0x7fe21180a000
mmap(0x7fe211810000, 14336, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7fe211810000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libz.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\320#\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=121280, ...}) = 0
mmap(NULL, 2216336, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fe211435000
mprotect(0x7fe211451000, 2097152, PROT_NONE) = 0
mmap(0x7fe211651000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1c000) = 0x7fe211651000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libxml2.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\320\354\2\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=1740024, ...}) = 0
mmap(NULL, 1747384, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fe21128a000
mprotect(0x7fe2112b8000, 1515520, PROT_NONE) = 0
mmap(0x7fe2112b8000, 1171456, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2e000) = 0x7fe2112b8000
mmap(0x7fe2113d6000, 339968, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x14c000) = 0x7fe2113d6000
mmap(0x7fe21142a000, 40960, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x19f000) = 0x7fe21142a000
mmap(0x7fe211434000, 2488, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7fe211434000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libdl.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0000\21\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=14592, ...}) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fe211288000
mmap(NULL, 16656, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fe211283000
[... cut ...]
write(2, " -b filter\t\t: filter output by l"..., 99 -b filter		: filter output by label (ex: -b "H1:Q0" : only output messages  with label H1 or Q0)

) = 99
write(2, "\nUp to 8 channels may be simulta"..., 48
Up to 8 channels may be simultaneously decoded
) = 48
exit_group(1)                           = ?
+++ exited with 1 +++
```
To be fair, I'm not sure I understand everything written here, but 
`access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (Aucun fichier ou dossier de ce type)`
is probably linked to my issue (no such file or folder)...

nsouleman, 2021-04-23:
Do you have any rtl_sdr usb stick plugged in?

gryzel1, 2021-04-23:
Forgot to plug it in before sending you the logs 😭
But actually, it seems like it doesn't change anything, like if the issue was happening before:
```
execve("/usr/local/bin/acarsdec", ["acarsdec", "-o1", "-v", "-p", "-22", "-g", "400", "-r", "0", "131.725"], 0x7ffd4f1cfc88 /* 49 vars */) = 0
brk(NULL)                               = 0x558cfa6ed000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (Aucun fichier ou dossier de ce type)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=119412, ...}) = 0
mmap(NULL, 119412, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7fa8e8112000
close(3)                                = 0
openat(AT_FDCWD, "/usr/local/lib/libacars-2.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\20\246\2\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=788240, ...}) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa8e8110000
mmap(NULL, 626936, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fa8e8076000
mmap(0x7fa8e80a0000, 180224, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2a000) = 0x7fa8e80a0000
mmap(0x7fa8e80cc000, 163840, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x56000) = 0x7fa8e80cc000
mmap(0x7fa8e80f4000, 114688, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x7d000) = 0x7fa8e80f4000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libpthread.so.0", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0@l\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=146968, ...}) = 0
mmap(NULL, 132288, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fa8e8055000
mmap(0x7fa8e805b000, 61440, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x6000) = 0x7fa8e805b000
mmap(0x7fa8e806a000, 24576, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x15000) = 0x7fa8e806a000
mmap(0x7fa8e8070000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1a000) = 0x7fa8e8070000
mmap(0x7fa8e8072000, 13504, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7fa8e8072000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libm.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0p\322\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=1579448, ...}) = 0
mmap(NULL, 1581384, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fa8e7ed2000
mmap(0x7fa8e7edf000, 651264, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xd000) = 0x7fa8e7edf000
mmap(0x7fa8e7f7e000, 872448, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xac000) = 0x7fa8e7f7e000
mmap(0x7fa8e8053000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x180000) = 0x7fa8e8053000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\260A\2\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=1824496, ...}) = 0
mmap(NULL, 1837056, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fa8e7d11000
mprotect(0x7fa8e7d33000, 1658880, PROT_NONE) = 0
mmap(0x7fa8e7d33000, 1343488, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x22000) = 0x7fa8e7d33000
mmap(0x7fa8e7e7b000, 311296, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x16a000) = 0x7fa8e7e7b000
mmap(0x7fa8e7ec8000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1b6000) = 0x7fa8e7ec8000
mmap(0x7fa8e7ece000, 14336, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7fa8e7ece000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libz.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\320#\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=121280, ...}) = 0
mmap(NULL, 2216336, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fa8e7af3000
mprotect(0x7fa8e7b0f000, 2097152, PROT_NONE) = 0
mmap(0x7fa8e7d0f000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1c000) = 0x7fa8e7d0f000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libxml2.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\320\354\2\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=1740024, ...}) = 0
mmap(NULL, 1747384, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fa8e7948000
mprotect(0x7fa8e7976000, 1515520, PROT_NONE) = 0
mmap(0x7fa8e7976000, 1171456, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2e000) = 0x7fa8e7976000
mmap(0x7fa8e7a94000, 339968, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x14c000) = 0x7fa8e7a94000
mmap(0x7fa8e7ae8000, 40960, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x19f000) = 0x7fa8e7ae8000
mmap(0x7fa8e7af2000, 2488, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7fa8e7af2000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libdl.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0000\21\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=14592, ...}) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa8e7946000
[... cut ...]
write(2, " -i stationid\t\t: station id used"..., 61 -i stationid		: station id used in acarsdec network format.
) = 61
write(2, " -b filter\t\t: filter output by l"..., 99 -b filter		: filter output by label (ex: -b "H1:Q0" : only output messages  with label H1 or Q0)

) = 99
write(2, "\nUp to 8 channels may be simulta"..., 48
Up to 8 channels may be simultaneously decoded
) = 48
exit_group(1)                           = ?
+++ exited with 1 +++
```

Maybe acarsdec can't find the driver, but rtl_test is working fine, using the same one.
I feel a little lost here haha

nsouleman, 2021-04-23:
Please make sure you installed these packages : librtlsdr-dev, librtlsdr0, rtl-sdr.
Use dmsg to check wether your stick has been detected by the system.
You should get something like : 
```
 rtl2832_sdr rtl2832_sdr.1.auto: Realtek RTL2832 SDR attached
[...]
 ```

gryzel1, 2021-04-23:
Tried auto-removing and installing these packages again, but nothing more happens...
The system detects the antenna as well:
`[16751.573502] rtl2832_sdr rtl2832_sdr.0.auto: Realtek RTL2832 SDR attached`
Maybe I should try on another machine

gryzel1, 2021-04-26:
Hi,
I just forgot to put `-Drtl=ON` while installing the software. I feel so dumb :(
Sorry for wasting your time and thank you for your help. It is working fine now.

closed 2021-04-26 by gryzel1