TLeconte/acarsdec #80: High CPU usage, no messages getting decoded

avgeeklucky, 2022-03-29:
After doing an update, I noticed no messages were getting decoded. Looking into it a little further I also noticed acarsdec was using 100% of a 32bit Intel atom CPU "core".

I've gone through the commits to try and find a culprit. 2ae8e8e847c2f4ab4ffb2e3c0bd78fb03fc0d319 seems to be the latest version that is working. From 2af3b865dc4287520daf7d68bfce8898fb621081 onwards it's 100% CPU usage and nothing gets decoded.

Undoing commit 2af3b865dc4287520daf7d68bfce8898fb621081 in current master (092989e389bfcdbd8bca0b94608f87b709c4a8e4) also seems to fix the problem.  I guess the old intel atom Does not like "portable" binary?

Other info: I'm using RTL-SDR as input device, and decoding 5 channels. OS is Debian Buster.

TLeconte, 2022-04-30:
#2af3b86 is very new. I committed it because some others have problems with -O2
"-Ofast -march=native" must works but the binary must be executed where it  have been compiled.

TLeconte, 2022-04-30:
I add a troubleshooting section to the README

closed 2022-04-30 by TLeconte