TLeconte/acarsdec #81: Using 2 instances of acarsdec on rpi

keukenrol, 2022-04-28:
Hi

I am interested in using 2 instances of acarsdec with two different rtl dongles. Reason being that the dongles have a bandwidth of 2.4MHz which is too narrow to cover both 131 and 136MHz acars channels.
What should be the best method? Two folders with acarsdec and their own configs?

Second question: Is it possible to have a web preview?

Kind regards

TLeconte, 2022-04-29:
- I never try that, but yes you must be able to run 2 acarsdec instances with different rtl device number (ie : -r 0 and -r 1 )
 But where will you send the results ? 
- What do you mean by a web preview ? An integrated web server ?
- You want that I add a csv format to -o option ?

TLeconte, 2022-04-29:
I will not code an integrated web server, not my domain.
Don't json a good format for the log ? Because it already exist : -o 4 
I don't think csv is a good format for acars because there are \n in message acars body
There is already the -l option to tell where acarsdec save its log

keukenrol, 2022-04-29:
Allright, fairly new to it all. I'll use the json. Thanks!

closed 2022-04-29 by keukenrol