TLeconte/acarsdec #82: Acarsdec on Raspberry pi with Bullseye 64 bit: too many parity errors but no decoded acars

keukenrol, 2022-04-29:
Hello

I recently installed acarsdec on a raspberry pi 3b+ with bullseye (raspbian 11) 64 bit.
I have the RTL-SDR blog V3 with good reception, However, until now I did not succeed in displaying decoded messages.
[image]
As you can see, it keeps displaying that there are parity errors, crc errors and that the errors cannot be fixed. This is in verbose mode. 
Am I missing something? I expected the decoded acars to be visible in the terminal with the -o 2 argument. How can I access the decoded data otherwise?

Many thanks!

TLeconte, 2022-04-29:
Having "too many parity erros" in verbose mode is normal , but not receiving messages is not.
-l option must be followed by a filename
Could you try : without -l and with less frequencies, because the more frequencies , the more CPU and I'm not sure of the oi 3b+ capabilities.

keukenrol, 2022-04-29:
Tried with 3 frequencies, both on raspberry pi 3b+ and 2b. Re-compiled and re-installed but same issue. No messages to see. Still using default 2 Ms/s. Only when using a single frequency it seems to work.

**Then recompiled with RTLMULT 80 (1 Ms/s) and now it works on the 3b+ with 3 channels.** 2b is still having issues with multiple.

TLeconte, 2022-04-30:
3B+ not powerful enough ? You could try to play  with the compile option in CMakefile.
If you have a libacars correctly  installed on your system, cmake will detect it and generate a Makefile that will use it.

closed 2022-04-30 by TLeconte