TLeconte/acarsdec #87: 136MHz supported?

keukenrol, 2022-05-11:
Hi

I have good reception on the 131MHz channels. I installed a new setup today to cover the 136MHz range, compiled the latest version from github. I notice however that no 136MHz messages are decoded. While the same setup does quite well on 131. Tried different gain settings without luck. Checked sdr# by streaming the spectrum and there is a lot of activity visible on the frequencies I scan. Any ideas?

fvance, 2022-05-12:
The 136 MHz data link channels are going to be VDL, not ACARS, so you need a different tool to decode them.

There are two I am aware of:
https://github.com/TLeconte/vdlm2dec
https://github.com/szpajder/dumpvdl2

keukenrol, 2022-05-12:
Excellent, I was not aware that is was a different message type. Thanks!

closed 2022-05-12 by keukenrol