TLeconte/acarsdec #98: Compile failure on Linux only "pthread_tryjoin_np" (rtl.c)

lawendel, 2022-09-13:
Bonsoir, 
Trying to compile on MacOS 12.6 on a M1 Air with clang 14, make generates the following error when encountering the rtl.c module: 

`/Users/lawendel/acarsdec/rtl.c:394:31: error: implicit declaration of function 'pthread_tryjoin_np' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
        while (count-- > 0 && (err = pthread_tryjoin_np(readThread, NULL))) {`

According to [GNU org] (https://www.gnu.org/software/gnulib/manual/html_node/pthread_005ftryjoin_005fnp.html) this function is missing on all non-glibc platforms. Before that I solved a few other error connected to libraries installed with Homebrew and other glitches, but I can't find an alternative for pthread_tryjoin_np

One solution was to compile for the Airspy, which aside from a single warning has generated an executable, still, it would be nice being able to use a RTL dongle as well!

Any suggestions?

Andy

lawendel, 2022-09-14:
I substituted pthread_tryjoin_np with pthread_join, which I've should give similar results. Now the executable has been compiled and installed, but it fails to recognize my rtl-sdr dongle. 

> lawendel@Air-di-Andrea ~ % acarsdec -r 131.725 131.825
> No matching devices found.
> Failed to open rtlsdr device
> Unable to init input

Analyzing the rtl.c file the error message seems to be connected to rtlsdr_open() function in rtl-sdr.h:

```
r = rtlsdr_open(&dev, dev_index);
	if (r < 0) {
		fprintf(stderr, "Failed to open rtlsdr device\n");
		return r;
	}
```
Is it a possible issue with this function?

Andy

TLeconte, 2022-09-14:
With rtl dongle you must set the dongle number before :
acarsdec -r 0 131.725 131.825

lawendel, 2022-09-14:
Merci! I had just found out, doh... All right, now it decodes beautifully, issue closed (looks like "pthread_join" works, or at least it doesn't give an error). I would like to track the received flights on a map, do you have a suggestion? With an AIS application using http.server I'm using OpenCPN, which should be compatible with JSON but I'm not sure it would work with Acarsdec generated format without any conversion. Planeplotter, I believe, is out of question...

TLeconte, 2022-09-14:
For flights on a map https://app.airframes.io/ will do that perhaps one day.
If you wanted plane position adsb is better suited.

closed 2022-09-14 by TLeconte

lawendel, 2022-09-14:
Didn't know about Airframes, it's an oustanding project! I'm using dump1090 with real time mapping for ADS-B, it's fascinating and it would be nice to have a similar interface for ACARS/VDL2 as well.
Thanks for your support, really.