TLeconte/acarsdec #103: -mfpu= Argument not recognised

keukenrol, 2023-02-09:
Hi

I have compiled version 3.7 on my raspberry pi 3b+.
When following the compilation guide in the readme, it states to change the add_compile_options line.
I have done this, but the -mfpu argument was not recognised. I did remove it and it compiled without further issues. Software works as expected.

best regards

Tim

TLeconte, 2023-02-10:
What is your gcc version ?
`gcc --version `

keukenrol, 2023-02-10:
gcc (Debian 10.2.1-6) 10.2.1 20210110

TLeconte, 2023-02-10:
Not too old.
That's strange because -mfpu is still documented :
[](https://gcc.gnu.org/onlinedocs/gcc-10.4.0/gcc/ARM-Options.html#ARM-Options)

0ip, 2023-02-17:
@keukenrol Are you running a 64 bit OS? I just had the same problem, removed that argument and it worked

keukenrol, 2023-02-17:
Yes, as commented I did the same thing to get it working.