TLeconte/acarsdec #113: trying to compile for sdrplay

sgoudelis, 2023-10-12:
Good evening,

I am trying to compile acarsdec in a ubuntu 22.04 and I am getting this: 

`
$ cmake .. -Dsdrplay=ON
-- Using libacars
-- Not using MQTT
-- Using libsnd
CMake Error at CMakeLists.txt:68 (message):
  libmirsdrapi-rsp path not found

-- Configuring incomplete, errors occurred!
`

I have the sdrplay api service install and running and I have a rspduo connected and already working well with openwebrx on it.

I can't find this library anywhere. Can someone point me to the right direction ?

McMaxFreak, 2023-11-30:
@sgoudelis I've got the same problem. As far as i know that's because API version 3.x does not provide the libmirsdrapi-rsp library. API version 2.x brings the libmirsdrapi-rsp along with it. The new API only has the libsdrplay_api.so file. I actually like the new name scheme much more then the old one. But still, that's the reason it doesn't compile.

If you found a workaround to get it working let me know.

Nocoderactually, 2023-12-28:
@sgoudelis @McMaxFreak  Same here! there is a repo [https://github.com/italic-r/mirsdrapi-rsp-sys](url) for the  mirsdrapi livrary but I have no idea how to install the thing.

Nocoderactually, 2023-12-29:
@sgoudelis @McMaxFreak  solved it by installing the 2.13 API from SDRplay. The newer versions don't run with this. It seems that the 2.13 API comes with _misdrapi_ whereas the newer versions don't have it